rumpeltux/esp-stlink #1: README should document host<->esp connection

spth, 2017-11-13:
This project might appeal to STM8 users unfamiliar with the ESP8266.  For those it would be helpful to document the connection between host and ESP8266 in the README. From the stm8flash fortk it seems the connection is done via a serial device. Is this done directly or somehow over the WIFI of the ESP8266? Is the WiFi used at all?

Philipp

rumpeltux, 2017-11-13:
Wifi is not currently used. It’s just using the serial device. Future version might include the functionality of this firmware in https://github.com/jeelabs/esp-link.
If you have an ESP dev board (e.g. NodeMCU or Wemos), it should work via USB out of the box.
I’ll add some more details for this to the README.

Re your P.S: I think this is covered by the "Connecting" section https://github.com/rumpeltux/esp-stlink#connecting If you don’t agree. What would you add?

spth, 2017-11-13:
I think the "Connecting" section currently only documents the connection between ESP8266 and STM8, not between ESP8266 and host.

Philipp

closed 2017-11-14 by rumpeltux