rumpeltux/esp-stlink #8: STM8L051 / lock at determine flash area

timactive, 2018-09-03:
Hi,
First of all thanks you for this amazing project.
I flash with success my esp.
When i try to flash my stm8l i have a message "determine flash area" but nothing is do after.
Have you any idea? can you assistate me to determine the error?
[image]

https://drive.google.com/file/d/1hw_O9Va9USuNkC7dnIcV-yU91pLdVZcv/view

timactive, 2018-09-03:
Hi Rumpeltux,
The commandline i use is :

```stm8flash -c espstlink -p stm8l051f3 -w blinky.ihx -s flash```

The output is only :
```Determine FLASH area```

2 The version of stm8flash is 20170616-1.1.

3 Nothing command work. same result.
Example of command i do for read 
stm8flash -c espstlink -p stm8l051f3 -r firmware.img

rumpeltux, 2018-09-03:
Can you give it a try with the current stm8flash HEAD version from https://github.com/vdudouyt/stm8flash?

rumpeltux, 2018-09-03:
Can you run
`strace -x stm8flash -c espstlink -p stm8l051f3 -w blinky.ihx -s flash`
and send me the output. It looks like it’s getting stuck somewhere but I don’t know where.

Also you can run `gdb stm8flash`
then
`r -c espstlink -p stm8l051f3 -w blinky.ihx -s flash`
Once it idles, hit `ctrl+c` and type `bt` and send me the output as well.

timactive, 2018-09-03:
Oh yes i am sure.
I change ESP 12 F to ESP 12 E.
I flash my ESP 12 E and i am not stuck to "determine flash".

```
[timactive@localhost sdcc-examples-stm8]$ stm8flash -c espstlink -p stm8l051f3 -w blinky.ihx -s flash
Determine FLASH area
Due to its file extension (or lack thereof), "blinky.ihx" is considered as INTEL HEX format!
71 bytes at 0x8000... Command 0x02 failed with code: 0x01
Command 0x01 failed with code: 0x03
OK
Bytes written: 71
```

```
[timactive@localhost sdcc-examples-stm8]$ stm8flash -cespstlink "-pstm8l051f3" -r flash.img
Determine FLASH area
Due to its file extension (or lack thereof), "flash.img" is considered as RAW BINARY format!
Reading 8192 bytes at 0x8000... Command 0x01 failed with code: 0x01

Requested 8192 bytes but received only 0.
Failed to read MCU
```

rumpeltux, 2018-09-03:
Some progress :) So now the communication with the espstlink works fine, but the SWIM communication doesn’t work reliably.
I’ve never tested with in stm8l, so there may be slightly different timing that causes the trouble or your esps clock speed may be different or there’s some clock skew.

You can try using different pullup resistor value. Also in your picture it looks like there’s a resistor between the GPIO header and the ESP pin? (Which breakout board is that btw?) You can try attaching directly to the ESP pin.

If all of this doesn’t work, further debugging requires an oscilloscope or at least a signal analyzer.

timactive, 2018-09-04:
Hi Rumpeltux,
Ok 
first of all I'll have to learn how to use an oscilloscope :-)
[image]

And have a little more perspective on the swim communication, I think it's a new adventure finally it remains a hobby.

closed 2018-09-04 by timactive