rumpeltux/esp-stlink #15: portable python driver

mrx23dot, 2021-08-24:
Is it possible to make \python\espstlink portable, (not depend on libespstlink.so) so it can be simply run on python only?

pyserial is an excellent portable serial lib instead.

mrx23dot, 2021-08-24:
pip3 install pyserial

```
import serial
seri = serial.Serial('COM6', 4000000, bytesize=8, timeout=0.1) 
# Clean start
if seri.isOpen():
  seri.close()
seri.open()

# give some time to actually open it, otherwise flush ineffective
sleep(0.1) 
seri.flush()
seri.reset_input_buffer() 
seri.reset_output_buffer()

seri.write(b'hello')
print(seri.read(10)) # 10 bytes or timeout in sec

```

rumpeltux, 2021-09-12:
I actually had an implementation of this already, though I don’t use it anymore. Added it for your reference. Pull-requests updating it are of course welcome.

commit d0226785 (2021-09-12): Add pure python implementation pyespstlink.py (not sure it still works though). Fixes #15
closed 2021-09-12 by rumpeltux