rumpeltux/esp-stlink #17: Binaries files cause esp rest

romeurolo, 2022-07-06:
hi, first thanks for all the work in this project

I try several time to flash the binaries from release with esptool, and never add success in the esp.
Before every try a erase the flash and also compare the binaries against what its in the flash and they are equal.

Just printing in the serial Monitor always:

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x3ffe8000, len 892, room 16 
tail 12
chksum 0x58
ho 0 tail 12 room 4
load 0x3ffe8380, len 932, room 12 
tail 8
chksum 0x64
load 0x40100000, len 26692, room 0 
tail 4
chksum 0x3d
csum 0x3d
rf_cal[0] !=0x05,is 0xFF

Sometimes changes after rst:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x3ffe8000, len 892, room 16 
tail 12
chksum 0x58
ho 0 tail 12 room 4
load 0x3ffe8380, len 932, room 12 
tail 8
chksum 0x64
load 0x40100000, len 26692, room 0 
tail 4
chksum 0x3d
csum 0x3d
rf_cal[0] !=0x05,is 0xFF

Im using esptool on Mac, thanks

rumpeltux, 2024-04-01:
Sorry, can’t really help about this one. Try a more stable power supply or googling for this symptom.

closed 2024-04-01 by rumpeltux

5ch4um1, 2024-04-20:
had a similar issue, reset loop, onboard led flashing, both with compiled and downloaded binaries:
` ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x3ffe8000, len 1304, room 16 
tail 8
chksum 0x14
load 0x3ffe8520, len 2848, room 0 
tail 0
chksum 0x5e
load 0x40100000, len 26268, room 8 
tail 4
chksum 0x47
csum 0x47
boot not set
ota1 not set
ota2 not set
system_partition_table_regist 2 success!!
rf_cal[0] !=0x05,is 0xFF`
forcing esptool in the Makefile to use the correct flash size for my module apparently fixed it for me:
`flash: $(FW_FILE_1) $(FW_FILE_2)
     $(ESPTOOL) --baud 460800 --port $(ESPPORT) write_flash --flash_size 4MB $(FW_FILE_1_ADDR) $(FW_FILE_1) $(FW_FILE_2_ADDR) $(FW_FILE_2)`