rumpeltux/esp-stlink #21: Doesn't Work, "Unexpected data" error

frak0d, 2024-03-09:
I have the same boards as shown in README, a Nodemcu with ch340 chip and a STM8S103F3P6. I have double checked the wiring. And yes i have the pullup resistor in right place.

I am now trying to flash a `blink.ihx` that I got from SDCC compiler.

But it keeps failing with `Unexpected data: xx` (replace xx with random hex values)

I have tried flashing with both `stm8flash` and `python/flash.py`, both giving the same error.
I also tried setting 115200 baud rate in `src/user/user_main.c` and in `libespstlink.c`, still same error in both flashing tools.
I also tried with and without pullup resistor, same error.
I also tried reducing delay in read_byte as suggested in #18 , still same error.

am i doing something wrong ?

rumpeltux, 2024-03-11:
It could also be that your power supply is not strong enough, what’s your host system?
And you do definitely need the pullup resistor.

frak0d, 2024-03-13:
Host is a Asus Tuf A15 laptop.

My setup is like:
Laptop USB Port --> NodeMCU --> stm8s103f3p6

the onboard micro usb of stm8 was not connected to anything.

rumpeltux, 2024-04-01:
Sorry, to further debug I’d probably attach this to a scope to see what’s going. Without that nothing I can do. If you figure it out, let me know or send a patch if it’s a software issue.

closed 2024-04-01 by rumpeltux

frak0d, 2024-04-03:
I got myself a STLINK-V2 Clone, but even with that [stlink](https://github.com/stlink-org/stlink) and  [stm8flash](https://github.com/vdudouyt/stm8flash) are unable to connect to SWD.

Only thing that worked for me was **ST Visual Programmer** (STVP) on windows.
(weirdly enough, it also fails to connect on first 2 tries, but then works fine after that)

Compiling in Arduino IDE and then flashing in STVP is inconvenient, but atleast it works.